skarab42/SLAcer.js #1: Problem slicing lattice-filled object

jmkao, 2016-03-26:
I setup up SLAcer.js with a 1920x1080 pixel resolution at 192mm x 108mm, and then imported the STL model. SLAcer.js laid out the model with the long part of the object along the X axis and the shortest axis up the Z axis. This is also the way that I laid out the model in Creation Workshop when I printed it previously.

The model seems to preview okay, but when I go to slice, it only makes it up to around layer 28 and then the window (in Google Chrome) hangs and I start to receive unresponsive page warnings from Chrome.

[Diffuser_Molds.zip](https://github.com/lautr3k/SLAcer.js/files/190378/Diffuser_Molds.zip)

lautr3k, 2016-03-26:
- The window slowdown problem is fixed (f1b78f60045c140448cacef662728619d225fde4) (too many geometries without faces).
- But Netfabb told that your files have errors. I think Creation Workshop has an auto repair tools.

For now I leave the issue open.
Thank you for the contribution.

jmkao, 2016-03-26:
Hmm... you are right. Let me generate the infill lattice a different way and test with that.

jmkao, 2016-03-26:
This one isn't lattice filled, but I'm also having some trouble with this calibration part:

https://github.com/CreativeTools/3DBenchy/raw/master/Single-part/3DBenchy.stl

There appear to be no non-manifold edges and vertices, but slices 75 and 80 contain triangular portions missing.

[image]

lautr3k, 2016-03-27:
Indeed, it is a bug. I need to investigate.

lautr3k, 2016-03-30:
I started to refactor all the code, but meanwhile a real fix an solution is to scale up or down a little bit (+/- 0.001 seems to work).

commit c4e5ce9c (2016-04-08): fix #1
closed 2016-04-08 by lautr3k