skarab42/SLAcer.js #5: Feature request: option to save ZIP locally on server or upload zip to server

rkoeppl, 2016-04-05:
For SLACER instandes running on an embedded system that controls the printer it would ake sense to store the sliced files on the server for further processing and printing. please add an option to either store locally or upload the ZIP after slicing.

lautr3k, 2016-07-22:
Done !  a2bedc9
Exported in slacer.json : 
{
  "imageExtension": ".png",
  "imageDirectory": "slices",
  "screenWidth": "640",
  "screenHeight": "480",
  "screenSize": "57",
  "screenUnit": "mm",
  "layersNumber": 100,
  "layersHeight": 0.1,
  "exposureTime": 4000,
  "liftingSpeed": 50,
  "liftingHeight": 4
}

jmkao, 2016-08-13:
We implemented a this in a fork, https://github.com/Kudo3D/SLAcer.js.

Take a look at the uploadZip() and makeButton() functions in https://github.com/Kudo3D/SLAcer.js/blob/gh-pages/js/photonic3d.js

A similar technique could be used for other embedded printer control systems (although I don't know of any open source ones other than Photonic3D).