skarab42/SLAcer.js #7: feature request: local printer control

rkoeppl, 2016-04-05:
For embedded systems it would be great to directly control the printer from the SLAcer menu.
ideally this would be done through a menu that allows to call local programs that interface with the printer and take the sliced part and the printer settings as parameter.
this would be especially great for Beaglebone black or Raspberry Pi running SLAcer and having a printer  + Beamer connected to GPIO and HDMI.

rkoeppl, 2016-05-07:
I have been working on this for some time now in my fork. As soon as initial testing is done I will send a pullrequest. Basically I have created a new folder to hold modules for different printer types and created a perl script to do the actual work and a configuration file for it. At this stage I am able to read in the configuration, do some sanity checks, hand a zip archive to the script as a command line option and define if that archive comes from SLAcer or some other tool. The zip is then extracted into a temporary folder, the png files sorted and sent to the projector. The part still missing is the interfacing with the stepper motor, the endstops and the like. I hope to add this in the next two weeks.
the code is being developed  and tested for a BeagleBone Black and a Lightcrafter 4500. It should however be easy enough to adapt to other platforms.

JamesNewton, 2016-08-09:
I'd like to second this request. But I would expect the output could be G-CODE sent via TCP/IP of some sort to a web enabled printer. E.g. via an ESP-8266 or other web to TTL serial shim. I don't see why that couldn't be generated directly via Javascript.